macformula/racecar #35: Build System: Add support for source files in mcal/shared folders

samparent97, 2024-01-17:
Currently all class definitions must be in header files.

BlakeFreer, 2024-02-05:
This will require the developer to add the required source files to the project-level CMakeLists.txt (like [this one](https://github.com/macformula/racecar/blob/main/firmware/projects/DemoProject/main.cc)). I support the change but want to note this.

BlakeFreer, 2024-03-18:
No changes to the build system are required. The code file can be added as a source in the project CMake file as follows:

```CMake
target_sources(main
PRIVATE
	main.cc
	${CMAKE_SOURCE_DIR}/shared/path/to/code.c
)
```

Note the use of `CMAKE_SOURCE_DIR` and not `CMAKE_CURRENT_SOURCE_DIR`. CURRENT refers to the directory containing the CMakeLists.txt file (in this case, `firmware/projects/DemoProject`), while `CMAKE_SOURCE_DIR` is the directory of the top-level CMake file, which is `firmware`, which is desired when locating shared code files.

closed 2024-03-18 by BlakeFreer