macformula/racecar #56: Improve ADC classes

BlakeFreer, 2024-02-27:
Make `Start()` private since only `Read()` should be called externally.

For stm32, constructor should take an `ADC_ChannelConfTypeDef` structure by value for configuration purposes.

BlakeFreer, 2024-08-26:
See #164

hsidd1, 2024-10-03:
The plan is to change the voltage level readings to be the actual voltage itself, right? To reduce overhead from using floating point everywhere, why not work with mV stored in uint32. For example, return 3300 (mV) rather than 3.3 (V) for a 2^12-1 reading on 12-bit ADC.

BlakeFreer, 2024-10-03:
Yes, we want to report the actual voltage. For example, with our 12 bit ADC, we return 4095 for 3.3V, 2047 for 1.65V etc. This issue is to perform the calculation at the `mcal` level so that the peripheral returns the voltage (3.3V or 1.65V etc).

I like your thinking about using `uint32`, but I think we should work with floats for a few reasons.
1. Floats use no more memory that uint32 (both are 32 bits)
2. We are nowhere near the computational limits of our processor, so we shouldn't worry about the performance cost of floats.
3. How would you perform the conversion from ADC to mV without going to floating point space? The conversion from [0-4095] to [0-3300] corresponds to a multiplication by 0.806. How will you implement that without converting to floating point anyways?
4. __Most importantly:__ An ADC is a peripheral which is used to measure a continuous signal. When we process a measurement from an ADC, we almost always perform floating point calculations on it. For examples, see the TMS temperature sensor class [firmware/projects/EV5/TMS/inc/app.h#L43](https://github.com/macformula/racecar/blob/main/firmware/projects/EV5/TMS/inc/app.h#L43) or the FrontController AnalogInput wrapper [firmware/projects/EV5/FrontController/inc/app.h#L30](https://github.com/macformula/racecar/blob/main/firmware/projects/EV5/FrontController/inc/app.h#L30) wrapper which

hsidd1, 2024-10-03:
So my idea was that we perform the floating point computation once in the read only then use uint32 to not need to deal with float elsewhere, but given the current usages, can stick to returning float. (Same idea as [this](https://stackoverflow.com/a/41632239))