macformula/racecar #76: pre-commit actions

BlakeFreer, 2024-03-22:


BlakeFreer, 2024-08-27:
We could have a clang-format (or better, git-clang-format) before any commit. See <https://pypi.org/project/clang-format/>

BlakeFreer, 2024-08-27:
Currently we have a GitHub pull request action that rejects a PR if the code is not clang-formatted.

This suggestion would execute `git-clang-format` locally whenever the developer makes a local commit. This ensures that all committed code is formatted before becoming a PR, which saves the developer time since they don't need the PR action to tell them to fix the code

Micnasr, 2024-09-06:
the .pre-commit-config.yaml has to be placed in the root of the directory from my understanding. The configuration works for me locally but it does not detect the file if its not in the root. What should I do?

BlakeFreer, 2024-09-06:
I think you should remove the `files=` line from the clang-format configuration. Use `git-clang-format` instead and let it figure out which files to change

BlakeFreer, 2024-10-12:
Closed by #172

closed 2024-10-12 by BlakeFreer