macformula/racecar #140: Update to STM32 IDE2, CLI tool, CMake generation

BlakeFreer, 2024-07-16:


Corbin-Jarrett, 2024-11-07:
I have been able to generate files using Cmake in STM32CubeMX. Some relevant differences between make:
1. The obvious difference, no Makefile is generated. Instead, a `cmake` directory is created.
2. This directory has a file `gcc-arm-none-eabi.cmake`
3. It also has a subdirectory `stm32cubemx` which contains `CMakeLists.txt`. I believe this file is a suitable replacement for `driver`

I have been able to get the build system to find and link the new driver library, now called `stm32cubemx`. I have reason to believe this is working because `bindings.cc` successfully builds. See attachment.
[image]

I am working through build errors with other files now. I am also unsure about how to go about regenerating only when `board_config` is updated. I will keep looking into this.

BlakeFreer, 2024-11-07:
> I have reason to believe this is working because bindings.cc successfully builds

This may not mean it is working since a lot of `bindings` relates to the linking process, so you'll know it worked once you see `linking bindings.cc` or something similar

> I am also unsure about how to go about regenerating only when board_config is updated

Don't worry about this. It can stay in a Makefile for now since _how_ we generate the code is less important than _what code_ is generated. You can assume the code is generated, your task is to integrate the generated CMake targets

PR #333 (2024-11-28): CubeMX with CMake
closes #140 

# Key changes

## CubeMX generates for CMake instead of Makefile!
- This greatly simplifies the build system since we don't need to rip data out of the Makefile.
- Also improves robustness since we can use the generated flags instead of fixed flags in Toolchain.cmake
- On that note, `toolchain.cmake` is no longer needed as it is generated on-the-fly by cubemx. No other platforms used it so I completely removed it
- CubeMX generation is executed in `mcal_conf.cmake` (instead of `platforms/stm32cubemx/CMakeLists`) because we need the toolchain to exist before CMak [... cut]
closed 2024-11-28 by BlakeFreer