macformula/racecar #165: Versioning our MCALs & build system

BlakeFreer, 2024-08-26:
Our repository has no versioning system which makes it near impossible to update our MCALs without breaking old projects.

- What would a versioning system look like?
- How would we store & compile against MCAL releases?
  - Should the MCAL / shared folder (and build system??) be moved to a separate repository from the projects?
  - How would these be integrated? 
  - Can CMake find the required release at build time?
  - :star: Could we use a Git Submodule checked out to a specific tag? (this may be the simplest / best)

samparent97, 2024-08-27:
My 2 cents here is that the old projects should evolve with the platform updates. Supporting legacy code is a consequence of a growing codebase. This is a big reason why CI will become important; we want to maintain the working order of older projects. 

One way to preserve these without requiring CI would be to make releases of the repo itself. Once working order has been achieved we create a repo release with release notes. Check out this [guide](https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository) on managing github releases.

BlakeFreer, 2024-10-02:
Closed for Sam's reason. Will use Repo Releases if necessary

closed 2024-10-02 by BlakeFreer