macformula/racecar #175: cangen does not work with some bus names

BlakeFreer, 2024-09-08:
`scripts/cangen` fails to generate code when the bus name contains lowercase letters.

TylerStAmour, 2024-09-09:
I couldn't reproduce this with lowercase or uppercase bus names in the config. Most of the configs contain lowercase bus names anyway. Could you share which project/dbc/config was giving you this issue?

BlakeFreer, 2024-09-10:
I meant to say "node names"

You are right about it working on most projects, but I'm having issues with `MotorDebug`

From inside `firmware/` run

```bash
python ../scripts/cangen/main.py --project EV5/debug/MotorDebug
```

Check the generated code in `MotorDebug/generated/can/messages. Works fine when when the bus name (in `pedal.dbc` and the projects `config.yaml`) is"FRONTCONTROLLER"

Change to "FrontContoller" in both files. Re-run the script and the generated code is empty

TylerStAmour, 2024-09-10:
Makes sense, I was able to recreate it now.

It looks like it's caused by the node name getting capitalized in the script, which mismatches the two names. When testing without uppercasing, it seems to generate correctly. The bus name also gets capitalized but it hasn't seemed to cause issues so far. Can we safely remove the `.upper()`?
https://github.com/macformula/racecar/blob/f115cbfd56be1a3128aac75a3ea048d58ca283f8/scripts/cangen/main.py#L53-L60

BlakeFreer, 2024-09-11:
@samparent97 Is there a reason why we need the nodes capitalized? Can we remove the `upper()` call on the node names? If so, let's do that. This is my preferred option

Otherwise we have 2 options:

1. Ensure that the node name is capitalized everywhere (ex whenever we read a node name from any file, automatically capitalize it.
2. Raise an error if the config uses lowercase letters

BlakeFreer, 2024-09-29:
We should not require node names to be capitalized.

AndrewI26, 2024-09-30:
It seems like this is not an issue anymore. The .upper() is necessary in order for the node name in the config.yaml of a project to match the name used in the .dbc file associated with that bus. Most of the node names have been written as lowercase, so when the CAN generator makes the generated files, it must receive the node name as uppercase. I tested it with and without the .upper() and it only worked with the .upper(). Maybe the naming of the motor debug was different in past versions of the repo but I think currently the .upper() is definitely needed.

BlakeFreer, 2024-09-30:
Reopening because I still have the issue on the current version of the repo.

@AndrewI26 The purpose of this issue is to enable lowercase node names in the `.dbc`. Currently, if `pedal.dbc` uses `FrontController` and `config.yaml` has `FrontController` then the generated file will have no messages, despite the names matching.

```c++
#pragma once

#include "shared/comms/can/can_msg.h"
#include "shared/comms/can/raw_can_msg.h"

namespace generated::can {

/* -------------------- ENUMS -------------------- */

/* -------------------- RX MESSAGES -------------------- */

/* -------------------- TX MESSAGES -------------------- */

// clang-format on

}  // namespace generated::can
```

If we remove the (somewhat arbitrary) `.upper()` call in `main.py` then this issue is not present.

The error is caused by `_filter_messages_by_node()` in `can_generator.py`. It isolates those messages whose node name _exactly_ matches the node name in `config.yaml`. We treat the node names differently by calling `.upper()` on the `.dbc` while not touching `config.yaml`. This means that identical strings in the file can become different from the perspective of the python script.

By removing `.upper()`, we compare both strings _as written_ in their respective files. Alternatively, we could define a function `normalize_node_name(str) -> str` which we call equally on both. This issue seeks the `upper()` solution, though we may implement the second in the future.

AndrewI26, 2024-09-30:
Ok sorry I think I understand the issue better now. I removed the .upper() call from the main script then added this code to the can generator file
```python
def _normalize_node_name(
    node_name: str
) -> str:
    return node_name.upper()

def _filter_messages_by_node(
    messages: List[Message], node: str
) -> Tuple[List[Message], List[Message]]:
    normalized_node_name = _normalize_node_name(node)
    tx_msgs = [msg for msg in messages if normalized_node_name in map(_normalize_node_name, msg.senders)]
    rx_msgs = [msg for msg in messages if normalized_node_name in map(_normalize_node_name, msg.receivers)]
``` 
This solution ensures that capitalization of node names in the config.yaml file and the .dbc file is irrelevant, since we capitalize both names anyways. Do you like this solution?

commit 3d85d31c (2024-09-30): fixed #175 CAN not working with lowercase (#195)
closed 2024-09-30 by BlakeFreer