macformula/racecar #206: RTOS on Linux / POSIX

BlakeFreer, 2024-10-02:
# RTOS on Linux / POSIX

## Description

Our firmware architecture is built on platform abstraction, meaning we can compile any project to run on any platform. This has huge benefits for debugging and development (see [this article](https://macformula.github.io/racecar/firmware/architecture/)) but is not fully implemented.

We have no way to build a FreeRTOS project for the CLI or SIL (raspi) platform, so we cannot test any project involving FreeRTOS. This is a huge problem since most of our EV5 projects run on FreeRTOS.

There are two ways which this project can be solved.

### First Solution: Port FreeRTOS

_This is the ideal solution but might not be possible._

Run FreeRTOS directly on Linux / Windows. This is the ideal solution because:

1. Using the same RTOS as we do on stm32 will allow us to remove the `shared::os` layer and allow us to use all of the FreeRTOS features.
2. We can expect other platforms to behave more like stm32 than if we use a different RTOS.

This may be impossible because FreeRTOS is fundamentally a real-time operating system that relies on thread preemption. This may not be available on a regular operating system.

## Second Solution: Custom Implementation

If FreeRTOS is not possible, we will emulate its functions using regular threads (tasks become threads, etc). Write an `mcal::cli::os` directory which implements the `shared::os` abstract classes using POSIX threads. Make a demo project showing that you can start tasks, synchronize with semaphores, and lock resources with mutexes.

## Decision

Research whether the first solution is possible. If you can't find anything, or if it would only work for linux and not windows, then proceed with the second solution.

TylerStAmour, 2024-10-02:
I just want to add one note about Linux, it has a new PREEMPT_RT patch that was just merged with the latest kernel version like 2 weeks ago. It adds soft realtime capability, maybe have a look there.

BlakeFreer, 2024-12-18:
>Make a demo project showing that you can start tasks, synchronize with semaphores, and lock resources with mutexes.

Expanding on this, here are some sample projects which would demonstrate a working OS layer:

# Demo 1 - Periodic Tasks (easier)

Create three tasks which print `Task 1/2/3` to the screen every 1, 2 or 3 seconds. Each task has an independent delay which doesn't block the other tasks. Something like

```c++
void Task1() {
    while (true) {
        print "Task1" + "time="current_time
        delay(1000 ms)
    }
}
// repeat for 2 and 3
```

Sample output: 

| time | output |
|---|------|
| 1| Task1 |
| 2| Task1, Task2 |
| 3| Task1, Task3 |
| 4| Task1, Task2 |
| 5| Task1 |
| 6| Task1, Task2, Task3 |
# Demo 2 - Synchronized User Input (harder)

(Do this after Demo 1)

Two buttons (simulated by keypresses) independently control two indicators (simulated by printing to the screen). Each button-led pair should run in a separate task.

Once this is working, simulate a "critical section" for each task: Use a mutex which only allows a single led to be on at any given time, so pressing one button will do nothing if the other is already pressed, and vice versa.

BlakeFreer, 2025-04-14:
https://freertos.org/Documentation/02-Kernel/03-Supported-devices/04-Demos/03-Emulation-and-simulation/Windows/FreeRTOS-Windows-Simulator-Emulator-for-Visual-Studio-and-Eclipse-MingW