macformula/racecar #324: pre-commit Error (file exists)

BlakeFreer, 2024-11-22:
I regularly get this error with the pre-commit clang format hook

```bash
don't commit to branch...................................................Passed
Run git-clang-format.....................................................Failed
- hook id: clang-format
- exit code: 2
- files were modified by this hook

`git read-tree --index-output=.git\clang-format-index --empty` returned 128
fatal: Unable to create 'C:/Formula/repos/racecar/.git/index.lock': File exists.

Another git process seems to be running in this repository, e.g.
an editor opened by 'git commit'. Please make sure all processes
are terminated then try again. If it still fails, a git process
may have crashed in this repository earlier:
remove the file manually to continue.
```

What is the cause of this? Has anyone else seen it or can reproduce it? I end up disabling pre-commit since I can't commit with it

Micnasr, 2024-11-23:
Hi Blake,
I tried reproducing the error but was unsuccessful. It seems to be working fine on my machine. I looked online and it seems that this can happen when you have another git process open like a second commit, or editor window opened by git. There could have been an issue regarding a previous git process which crashed leaving a lock file active. This can be deleted using `rm -f .git/index.lock`

People also mentioned that I could add the `--force` parameter to bypass this error.

Could you try to see if this works with the new changes as I cannot get this error.

Thanks

Micnasr, 2025-01-12:
Hi, I was wondering if you still get this error

BlakeFreer, 2025-01-12:
> Hi, I was wondering if you still get this error

Yes, I get this error and @ManushPatell had it as well. There's no point working on this if you can reproduce it

BlakeFreer, 2025-02-24:
This might be a problem with the VSCode GitGraph extension. Issue https://github.com/mhutchie/vscode-git-graph/issues/583