macformula/racecar #355: DBC to JSON

BlakeFreer, 2024-12-26:
### Discussed in https://github.com/macformula/racecar/discussions/275

<div type='discussions-op-text'>

<sup>Originally posted by **BlakeFreer** November  1, 2024</sup>
What might this look like? How and where will it be parsed? Are there existing standard formats?</div>

BlakeFreer, 2024-12-26:
UBC uses JSON for their CAN database. Again, this issue is not about JSON vs YAML, but you can see how they chose to structure their database. I'm not endorsing it in any way, but it may be a source of inspiration

BlakeFreer, 2024-12-26:
Check out the `racecar/datasheets/amk` for a description of the AMK motor CAN messages. Our database obviously needs to support those motor messages.

luaibash, 2025-01-04:
Here are my initial thoughts with reading this issue:

- `@dataclass` definitely does help reduce boilerplate code, so let's go ahead with using that for defining the classes.
- UBC's [CAN database implementation](https://github.com/UBCFormulaElectric/Consolidated-Firmware/tree/master/scripts/code_generation/jsoncan) is pretty useful for understanding what we want to do and what it could look like.

Responses to questions:

1. **Should we have two separate databases:** I think we should keep these separate, as it follows better coding practice.
2. **Which parameters should be optional:** I don't see any parameters that should be optional yet, but what I'm thinking with this is that we set optional parameters that have default values if the value is not provided by the instance.
3. **The message receivers are specified per signal. Does this make sense:** I think it's a good idea to try and "set the receiver(s) for the entire message to simplify the database". Taking a quick look at the messages in [pt.dbc](https://github.com/macformula/racecar/blob/main/firmware/projects/pt.dbc), it looks like that would be the simpler implementation as all signals within the same message use the same receiver. We just have to ask ourselves, could there ever be a scenario where a message could have multiple receivers? If no, is it something we want to support for the future?

BlakeFreer, 2025-01-04:
> could there ever be a scenario where a message could have multiple receivers?

This is very possible. The TMS temperature message is used by the BMS and FrontController. But is there ever a message for which some (say 5 of 8) signals are used by a node and other (3 of 8) signals aren't? I don't think so, so let's specify a list of receiver nodes per message, not per signal as dbc does

Jinwoo-H, 2025-01-07:
Here are some of my thoughts

I agree we should keep the two databases separate

I also don't think there are any params that should be optional (excluding params that are exclusive to integer vs floating signal types), but we could have an optional `description` or `comments` params for better readability

Even if currently all signals in a message are received by the same node(s), I think it would be good to have the granularity of specifying receivers for different signals if that is ever required in the future, as well as the benefit of staying consistent to how DBC works. I'm very open to discuss though.

Either way, I think a set makes more sense for the `receiver` attribute to be a set (as opposed to a list)

- There shouldn't be duplicate receivers. A list will allow duplicates which could mean it is possible for a signal to be received twice if not handled properly

- It's a quicker/ more straightforward lookup when checking if a signal/message should be received by a certain node

I do think the `signals` attribute should be a list though

- It is probably important to maintain the ordering of signals which unordered sets don't do
- It doesn't have to same constraints as the receiver attribute where a quicker lookup time is beneficial
- If we want to preserve uniqueness of signals, we could use an ordered set if necessary, but otherwise I think a list would suffice

I agree that having a signal type param would be useful, it can look something like this
```
class SignalType(Enum):
    INTEGRAL = auto()
    FLOATING = auto()
```

To specify different enums across signals, it can look something like this
```
@dataclass
class EnumValue:
    name: str
    values: Dict[int, str] # integer and its corresponding meaning/name
```

BlakeFreer, 2025-01-07:
> To specify different enums across signals, it can look something like this

```
@dataclass
class EnumValue:
    name: str
    values: Dict[int, str] # integer and its corresponding meaning/name
```

We shouldn't implement bare enums ourselves. Python has "functional syntax" for defining enums at runtime (see official docs). My question is more about _where_ enums should be defined in the data structure, not how

BlakeFreer, 2025-01-07:
> I do think the signals attribute should be a list though

@Jinwoo-H Do you mean "I __don't__ think"? If so, I agree since there is a natural order to the signals

Jinwoo-H, 2025-01-08:
> > I do think the signals attribute should be a list though
> 
> @Jinwoo-H Do you mean "I **don't** think"? If so, I agree since there is a natural order to the signals

I meant I think we should have signals as a list, as a sets in python will not preserve the ordering of the signals so we must use a list instead, but I think I understand what you mean about the signals already have a natural order that we can follow instead, which in that case we can just use a set

BlakeFreer, 2025-01-08:
@Jinwoo-H sorry I mistyped. Yes we should have signals as a list.