macformula/racecar #370: Build FrontController for CLI Platform

Tegh25, 2025-01-05:
The `FrontController` project should be buildable for the cli platform so that classes, controls, etc. can be tested with their own unique test cases in a easily debugged environment.

The current workaround for this is creating a new, separate project for the subsystem that needs testing. This involves creating cmake and bindings configurations, which is inefficient.

Reference below:
- https://github.com/macformula/racecar/pull/369
- https://github.com/macformula/racecar/pull/330

BlakeFreer, 2025-01-06:
This would be really great the have. The FrontController uses FreeRTOS which we haven't ported to the `cli` platform yet. See #206. Alternatively, if RTOS will be too difficult to get working for other platforms, then we could remove the RTOS and manage the FrontController scheduling on our own.

We will need one of these solutions since the FrontController will need to support the `sil` platform for testing.

BlakeFreer, 2025-02-24:
Accomplished in https://github.com/macformula/racecar/pull/400

closed 2025-02-24 by BlakeFreer