macformula/racecar #373: `pre-commit` warning

BlakeFreer, 2025-01-08:
When running `pre-commit install` get a warning "stage name `commit` will be removed.

# Task

Follow the instructions in the warning message to migrate our precommit file

Micnasr, 2025-02-05:
Not sure if I'm doing this right but this is what I've been getting when running the command.

`PS C:\Users\Michael\OneDrive\Documents\FormulaElectric\Repos\racecar\firmware> pre-commit uninstall
pre-commit uninstalled
PS C:\Users\Michael\OneDrive\Documents\FormulaElectric\Repos\racecar\firmware> pre-commit install  
pre-commit installed at .git\hooks\pre-commit`

BlakeFreer, 2025-02-24:
You're right, I'm not getting this error anymore. Not sure what that was. Closing the issue

closed 2025-02-24 by BlakeFreer