macformula/racecar #383: Is the `stm32` platform generic?

BlakeFreer, 2025-01-14:
Can we use the current `stm32f767` platform to compile for other `stm32f` microcontrollers? If not, what is stopping us?

This would be awesome to have since Dashboard uses a different but similar chip, so reusing the mcal means we would have to rewrite it. 

Since we use the stm HAL, most of our code should be portable. Is there anything non portable? A good first check would be searching "767" in the `firmware/mcal` folder. 

Further, is it possible to generate CubeMX code for two different boards and have them fit into the same PROJECT-PLATFORM?

> Prior to #333, the answer was NO since we hard-coded the `mcpu` compiler flag into our build system. That PR improved it to use the generated flags, opening up this possibility.

BlakeFreer, 2025-01-31:
Almost but not exactly as written. Tried to use the stm32f767 mcal with the f4 chip on the Dashboard but they use different `stm32f7xx_hal.h` and `stm32f4xx_hal.h` files. Also f4 only has CAN1, CAN2 not CAN3 so the can driver needs to be changed.

Instead of making a duplicate mcal folder, I think we should use the preprocessor to enable certain sections of code depending on which processor is being used

Tegh25, 2025-04-06:
I am able to build a project for the `stm32f4` platform using the `Demo/Blink` project and a *mostly* untouched `stm32f767` mcal. After building a cubemx project using the [documented instructions](https://macformula.github.io/racecar/firmware/stm32/start-cubemx-project/), I made the following changes:
1. Replacing the f7 header with `stm32f4xx_hal.h` in `gpio.hpp` and `can.cc` within the `\firmware\mcal\stm32f767\periph\` directory.
2. Copied `bindings.cc` from `stm32f767`, but changing the GPIO pins to match the ones enabled on the f4.

The project/platform can then be built successfully using `make PROJECT=Demo/Blink PLATFORM=stm32f4 build`

The next step should be to implement macros in the mcal that identify the platform being built and include the appropriate stm HAL.

BlakeFreer, 2025-04-06:
You don't need to implement those macros on your own. STM defines them for us. Take a look in the cubemx generated code. You should see things like

```c
#ifdef __STM32F4__
    #include "f4.h"
#endif

#ifdef __STM32F7__
    #include "f7.h"
#endif
```

We just want to do that same thing with whichever parts of our mcal are chip-dependent

Tegh25, 2025-04-06:
Added the following to the chip dependent files in the f7 mcal:
```c
#ifdef STM32F7
    #include "stm32f7xx_hal.h"
#elif defined(STM32F4)
    #include "stm32f4xx_hal.h"
#endif
```

I have only tested it with the `stm32f767` and `stm32f4` platforms in the Demo/Blink project so far, which use `gpio.hpp` and `can.cc`. I am able to build both platforms without having to make any changes to the mcal.
I pushed my changes to the [mcal-platform-comp branch](https://github.com/macformula/racecar/blob/user/atelieyt/mcal-platform-comp/firmware/mcal/stm32f767/periph/gpio.hpp#L10) so that you can see what I've done.

BlakeFreer, 2025-04-06:
Can you start a draft PR? Then I can see and comment on it

Tegh25, 2025-04-07:
@BlakeFreer draft PR can be found [here](https://github.com/macformula/racecar/pull/446).

closed 2025-04-16 by BlakeFreer